M17-Project/rru-rf-hw #8: Sampling rate issue

sp5wwp, 2023-12-18:
MCU has to know when to send/read baseband samples from the transceivers. Now, the sample clock is run at the MCU side at 24kHz, and causes occasional sample drops and desynchronization.

Solution:
Connect GPIO3 pins of both CC1200s to the MCU's EXTI pins and let TRXs dictate the baseband transfer speed.

sp5wwp, 2023-12-19:
New firmware has a `#define` to handle the bodge wire mod.

ferdem, 2023-12-19:
unsubscribe

Wojciech Kaczmarski ***@***.***>, 19 Ara 2023 Sal, 16:44
tarihinde şunu yazdı:

> New firmware has a #define to handle the bodge wire mod.

sp5wwp, 2024-01-05:
The new `rpi-interface` branch supports the bodge wire mod without the need of having any #defines.

MCU passes baseband samples to the transmitter at the programmed rate (1.5*16=24kHz), trigger signal is provided by the CC1200. The receiver's `CLKEN_CFM` clock signal is always about 75.7582kHz, regardless of the sample rate setting. Baseband samples from the receiver are therefore read at a rate dictated by TIM7.

sp5wwp, 2024-01-18:
TX sample clock fixed in 636a110.

closed 2024-01-18 by sp5wwp