M17-Project/rru-rf-hw #18: License question

sedlund, 2024-06-26:
I see this project is using the 'CC Attribution-NonCommercial-ShareAlike 4.0 International' license.  This is not an OSI approved Open Source license.

M17 project on it's web page says it is an open source project.

How was this license chosen for this project? 

When someone donates to M17 can they choose to only fund open source projects?  Is this documented anywhere?

sp5wwp, 2024-06-26:
Thanks for your comments.

>This is not an OSI approved Open Source license.
It doesn't mean that it's not open source. Is the documentation/code available for free to anyone? Yes - for me it's open source.

>M17 project on it's web page says it is an open source project.
Because it is. All our work is available to the general public for free.

>How was this license chosen for this project?
I picked one I liked the most at that time. Since I'm the sole author, I can change them in the future however I desire.

>When someone donates to M17 can they choose to only fund open source projects? Is this documented anywhere?
When donating, the donor gives us a certain amount of money, to sustain the *whole* Project. I don't think our fiscal sponsor allows for funds earmarking per donation.

sp5wwp, 2024-06-26:
`Open source is source code that is made freely available for possible modification and redistribution.`

Which part of this sentence do we violate?

sedlund, 2024-06-26:
Specificlaly the current license states:

```
Section 2 -- Scope.

  a. License grant.

       1. Subject to the terms and conditions of this Public License,
          the Licensor hereby grants You a worldwide, royalty-free,
          non-sublicensable, non-exclusive, irrevocable license to
          exercise the Licensed Rights in the Licensed Material to:

            a. reproduce and Share the Licensed Material, in whole or
               in part, for NonCommercial purposes only; and

            b. produce, reproduce, and Share Adapted Material for
               NonCommercial purposes only.
```

This violates the requirements of the open source definition against discrimination of fields of endeavor - parts 5 and 6 of the [open source definition](https://opensource.org/osd).

sp5wwp, 2024-06-26:
OK: 5. is not violated, while 6. is, deliberately (non-commercial clause). This is exactly how I wanted it to be. Do you have any further comments or questions?

sedlund, 2024-06-26:
I appreciate the clarification.